modm-io/modm #1224: DynamicArray test failure due to no-op vector::shrink_to_fit implementation

WasabiFan, 2024-11-11:
After the switch to use std::vector for DynamicArray (fdb858289e8006b062907435eb576ddccd0575e0), I have the following test failure on my STM32H753 (custom BSP):

```
>>> dynamic_array
FAIL: dynamic_array:173 : 5 == 0
```
The test does a `clear()` and expects the capacity to have reset to 0. However, the internal shrink_to_fit() call had no effect and the storage remains allocated.

It looks to me like the gcc libstdc++ implementation of shrink_to_fit is hard-coded to `return false` unless you enable C++ exception support. Indeed, if I set `modm:stdc++:exceptions=yes`, the tests all pass.

cppreference says the following on [vector::shrink_to_fit](https://en.cppreference.com/w/cpp/container/vector/shrink_to_fit):

> `std::vector<T,Allocator>::shrink_to_fit`
> 
> Requests the removal of unused capacity.
>
> It is a non-binding request to reduce capacity() to size(). It depends on the implementation whether the request is fulfilled.

So it seems this behavior is acceptable for the C++ std library implementation.

What's the appropriate resolution here? If clear() is expected to unconditionally reclaim the unused space, I'm not sure that can be done without customizing the std implementation. Alternately, if it's acceptable for clear() to _not_ reclaim the space, then I suggest we remove the failing assertion.

salkinium, 2024-11-11:
> Alternately, if it's acceptable for clear() to not reclaim the space, then I suggest we remove the failing assertion.

Yes. There's nothing we can do anyways, short of reimplementing `std::vector`, so let's remove that test.

chris-durand, 2024-11-11:
It can be fixed. Instead of calling `shrink_to_fit()` swapping the vector with a copy of itself  with really shrink it, even with exceptions turned off: `std::vector(data).swap(data);`
This is an old pre-C++11 trick that was used before `shrink_to_fit()` was a thing. I'll open a PR tomorrow.

PR #1226 (2024-11-12): [container] Fix reducing capacity to 0 on DynamicArray::clear()
Fixes #1224

@WasabiFan Could you check if the test passes with that fix?