modm-io/modm #1228: nucleo-l452-re lvgl example wrong display buffering mode

frnktank, 2024-11-19:
The example for rendering on the ili9341 via lvgl in examples/nucleo_l452re/lvgl/main.cpp uses the direct draw mode for d´te display, but the buffer is not screen sized. Quote from the [documentation](https://docs.lvgl.io/9.2/porting/display.html): 
> " LV_DISPLAY_RENDER_MODE_DIRECT: The buffer(s) has to be screen sized"

I think the mode should be partial for this buffer as a screen sized buffer could be to large for onboard memory.
Also had an issue with the buffer being not aligned, adding modm_aligned(4) as attribute helped.

salkinium, 2024-11-20:
That's my fault, I upgraded the [LVGL from v8 to v9](https://github.com/modm-io/modm/pull/1211) and didn't have that specific hardware to test it on, only the DISCO-F469 (see the missing checkbox).
Please open a PR, since I cannot fix it!

frnktank, 2024-11-22:
PR should be opened now.

closed 2024-11-22 by salkinium