modm-io/modm #1250: Missing `Rcc::enableInternalClockMHz48` Function for Some STM32 Microcontrollers

dokee39, 2025-01-02:
I am using the STM32G431KBT6 microcontroller and would like to enable HSI48 for the USB peripheral. However, I found that only the function `Rcc::setClock48Source(Rcc::Clock48Source::Hsi48)` is available, but there is no corresponding function to enable HSI48.  

I noticed that the library does include the function `Rcc::enableInternalClockMHz48`, but it is not generated in my code.

https://github.com/modm-io/modm/blob/ab7c1ca29a2bcc47e02ba1a991b6e138549d5303/src/modm/platform/clock/stm32/rcc.cpp.in#L44

salkinium, 2025-01-02:
The inclusion depends on this query, could you play around with that and open a PR if it works for you?
https://github.com/modm-io/modm/blob/ab7c1ca29a2bcc47e02ba1a991b6e138549d5303/src/modm/platform/clock/stm32/module.lb#L71-L75

dokee39, 2025-01-02:
I see, thanks for explaining! I’ve never created a PR before, but I’ll give it a try.

PR #1252 (2025-01-05): [stm32] add missing HSI48-related functionality
Fixes #1250

- add `Rcc::enableInternalClockMHz48` function for stm32 g0 g4 l0 h7 (h5 u0)
- make changes to `PllSource`, `SystemClockSource`, `Clock48Source` and `UsbClockSource`

| family | hsi48_cr |
|:-------|:---------|
| F0 (42, 48, 71,72,78, 91, 98) | CR2 |
| G0 (B1 C1) | CR |
| G4 | CRRCR |
| L0 (x2) | CRRCR |
| L4 (`target["name"][0] not in ["7", "8"]`) | CRRCR |
| U0 | CRRCR |
| L5 | CRRCR |
| U5 | CR |
| H5 | CR |
| H7 | CR |
closed 2025-01-05 by salkinium