modm-io/modm #1259: Cannot generate docs with Doxypress

hamkoroke, 2025-03-12:
Hi all, I just started using this project. It is so promising and exciting. Thank you all for this.
Hope I can start using it more and send more exciting issues or questions.

Just wanted to ask if someone gets this issue or it is only my setup/version problem.

While executing `doxypress doxypress.json` under the generate `modm/docs` directory the generation process gets stuck and never ends:

```
disc_board/modm/docs$ doxypress doxypress.json 

DoxyPress Version: 1.7.0

**  Read Project Configuration
**  Verify Project Configuration

Initialization
Parse input files

**  Generate Documentation Output <--- never goes beyond this point
```

So I run it with GDB and sent a SIGINT to see more or less what was doing:

```
Program received signal SIGINT, Interrupt.
0x0000555555b7e8fd in CsString::CsBasicString<CsString::utf8, std::allocator<unsigned char> >::operator[](long) const ()
(gdb) bt
#0  0x0000555555b7e8fd in CsString::CsBasicString<CsString::utf8, std::allocator<unsigned char> >::operator[](long) const ()
#1  0x0000555555f2be5f in removeRedundantWhiteSpace(QString8 const&, bool) ()
#2  0x0000555555f35cc9 in getResolvedClassRec(QSharedPointer<Definition const>, QSharedPointer<FileDef const>, QString8 const&, QSharedPointer<MemberDef>*, QString8*, QString8*) ()
#3  0x0000555555f373ed in getResolvedClass(QSharedPointer<Definition const>, QSharedPointer<FileDef const>, QString8 const&, QSharedPointer<MemberDef>*, QString8*, bool, bool, QString8*) ()
#4  0x0000555555f3c3dc in getDefs(QString8 const&, QString8 const&, QString8 const&, QSharedPointer<MemberDef>&, QSharedPointer<ClassDef>&, QSharedPointer<FileDef>&, QSharedPointer<NamespaceDef>&, QSharedPointer<GroupDef>&, bool, QSharedPointer<FileDef>, bool, QString8 const&) ()
#5  0x0000555555f99d37 in getLinkInScope(QString8 const&, QString8 const&, QString8, CodeGenerator&, QString8 const&, bool) ()
#6  0x0000555555f9a7d5 in getLink(QString8 const&, QString8 const&, CodeGenerator&, QString8 const&, bool) ()
#7  0x0000555555f9f01e in generateClassOrGlobalLink(CodeGenerator&, QString8 const&, bool, bool) [clone .constprop.0] ()
#8  0x0000555555fb09ed in code_cstyle_YYlex() ()
#9  0x0000555555fb25e9 in parseCCode(CodeGenerator&, QString8 const&, QString8 const&, SrcLangExt, bool, QString8 const&, QSharedPointer<FileDef>, int, int, bool, QSharedPointer<MemberDef>, bool, QSharedPointer<Definition>, bool) ()
#10 0x0000555556016f70 in CPP_Parser::parseCode(CodeGenerator&, QString8 const&, QString8 const&, SrcLangExt, bool, QString8 const&, QSharedPointer<FileDef>, int, int, bool, QSharedPointer<MemberDef>, bool, QSharedPointer<Definition>, bool) ()
#11 0x0000555555cf8735 in FileDef::parseSource(bool, QStringList&) ()
#12 0x0000555555b4ca36 in Doxy_Work::generateSourceCode() ()
#13 0x0000555555b78c58 in generateOutput() ()
#14 0x0000555555ae4cd9 in main ()
```

This is my project.xml:

```xml
<library>
  <repositories>
    <repository><path>../../modm/repo.lb</path></repository>
  </repositories>
  <options>
    <option name="modm:target">stm32f407vgt6</option>
  </options>
  <modules>
    <module>modm:board:disco-f407vg</module>
    <module>modm:docs</module>
    <module>modm:build:compilation_db</module>
    <module>modm:build:scons</module>
  </modules>
</library>
```

I'm using modm version at `2024q4`.

Thank you again.

hamkoroke, 2025-03-12:
I understand that this issue is from doxypress but I wanted to report it because the documentation is one of the best parts of this project and I thought it is important to mention. Maybe other users face same issue and cannot access documentation as well.

salkinium, 2025-03-12:
Thanks for the kind words!

Try using Doxypress v1.4.2. That's what works in our CI. The newer versions have a lot of stability and speed issues sadly.

closed 2025-03-15 by hamkoroke