modm-io/modm #1277: RTT buffer size computation incorrect for empty buffer

WasabiFan, 2025-07-16:
I see this getSize implementation has gone through a few iterations based on commit history. It seems to me there's still a bug.

https://github.com/modm-io/modm/blob/3fbe3a1ab2c3c30c3a69aa5b2a657f11780bfe1f/src/modm/platform/uart/rtt/rtt.cpp.in#L47-L53

On construction:
- head = 0
- tail = 0
- getSize() returns `MAX + 0 - 0 = MAX`

After writing one byte:
- head = 1
- tail = 0
- getSize() returns `0 + 1 - 0 = 1`

My best interpretation for the intent of this method is to return the number of occupied bytes. If so, the behavior for an empty buffer is incorrect.

I think the implementation in the submitted version of #610 is correct, with one caveat. If `head == tail`, it is ambiguous whether the buffer is full or empty. I suspect baa88d24997346d60d11f4f6c7276a1d85f8a8ac was attempting to work around this limitation. I think the change moves the bug but doesn't fix it.

I am not sure what the RTT format expects for boundary conditions. One entry must be reserved to disambiguate unless there's additional state (e.g. entry count). The original implementation works so long as one entry is reserved. If that's the expected behavior, isEmpty() and write() would need to be updated to match.

salkinium, 2025-07-20:
TIL: there is [an official wiki for RTT](https://kb.segger.com/RTT).

>  J-Link will first try to locate the Control Block's address in the vector table at offset 0x20. 0x2 should be added to the address to mark it as a valid RTT Control Block address.

Interesting, I should implement this.

I think here's the official implementation of the getSize() function: https://github.com/SEGGERMicro/RTT/blob/master/RTT/SEGGER_RTT.c#L523

chris-durand, 2025-07-20:
> I think here's the official implementation of the getSize() function: https://github.com/SEGGERMicro/RTT/blob/master/RTT/SEGGER_RTT.c#L523

Wasn't getSize() rather meant to be [SEGGER_RTT_GetBytesInBuffer()](https://github.com/SEGGERMicro/RTT/blame/ff57c3d516ea59b0a2d4d5cafb33a48572b09c0b/RTT/SEGGER_RTT.c#L2069)? Then the implementation in #610 would be correct as @WasabiFan stated and the fix in [ba88d2](https://github.com/modm-io/modm/commit/baa88d24997346d60d11f4f6c7276a1d85f8a8ac) broke it for the case of an empty buffer.

I suspect the confusion stems from the fact that the expected semantics of a `getSize()` function are different when you look at either the receive or transmit case. You'd either care about the amount of the occupied bytes that you can read or the free space available for writing. We should probably add two functions with more descriptive names to cover both use cases.

We seem to have missed a few more details. Segger [inserts a `DMB` barrier](https://github.com/SEGGERMicro/RTT/blame/ff57c3d516ea59b0a2d4d5cafb33a48572b09c0b/RTT/SEGGER_RTT.c#L409) before updating the `head` pointer in the write case because a Cortex-M7 or Arm v8 core could reorder the pointer update with the data write.

salkinium, 2025-07-20:
Meh, the Segger RTT code is BSD licensed, so instead of thinking about properly, I'm just going to instead integrate their code into modm 😝.

chris-durand, 2025-07-20:
@salkinium I've found something odd in the Segger RTT wiki:

> Cortex-M specifics
>
>If the CPU implements caches:
>    - The application must provide a uncached address alias to the memory where the control block + buffers are located in. The control block as well as all buffers need to be accessible via that uncached address alias.

I don't think you can actually do that on Cortex-M without an MMU. I guess the RTT buffers must be placed in non-cacheable memory then?

salkinium, 2025-07-20:
Yup saw that too. There are too many things are wrong with our implementation.

We would need a non-cachable memory region anyways for the DMA issues we have.

closed 2025-08-09 by salkinium

WasabiFan, 2025-08-17:
Thanks to both of you! I was out of town for a few weeks but this looks great and I will give it a try.

PR #1278 (2025-08-09): Integrate SEGGER RTT implementation
- [x] Add SEGGER RTT code as submodule
- [x] Add `modm:rtt` module with modm port.
- [x] Refactor `modm:platform:rtt` to use the SEGGER RTT code.
- ~~Figure out a way to place all buffers into uncached memory (using the MPU).~~ doing this later
- [x] Update all related documentation
- [x] ~~Fix telnetlib3 API change~~ Switched to using netcat, much better.
- [x] Test in Hardware 
  - [x] on STLink
  - [x] on JLINK

Closes #1277 by definition.

cc @WasabiFan @chris-durand